0neblock/Arduino_SNMP #3: addOIDHandler

AdamPiorko, 2018-07-28:
Will there be an addOIDHandler function added?
I need to return information on question 1.3.6.1.2.1.1.2.0 which requires the OID data type.

For example:
# snmpget -v 1 -c public <IP> 1.3.6.1.2.1.1.2.0
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.14988.1

0neblock, 2018-07-30:
Hey,

I made a quick addition that enables this in commit #4b7f89ef
snmp.addOIDHandler(".1.3.6.1.2.1.1.2.0", ".1.3.6.1.4.1.14988.1");

Let me know if it works.

AdamPiorko, 2018-07-31:
Something is wrong.
snmp.addOIDHandler(".1.3.6.1.2.1.1.2.0", ".1.3.6.1.4.1.**14988**.1");
responds
snmpget -v 1 -c public 192.168.2.186 1.3.6.1.2.1.1.2.0
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.**14988**.1

snmp.addOIDHandler(".1.3.6.1.2.1.1.2.0", ".1.3.6.1.4.1.**52421**.1");
responds
snmpget -v 1 -c public 192.168.2.186 1.3.6.1.2.1.1.2.0
iso.3.6.1.2.1.1.2.0 = OID: iso.3.6.1.4.1.**3269**.1

0neblock, 2018-08-01:
Hey, 
I noticed this too, it was a problem with the way I was encoding the ASN.1 packets.
Fixed in commit: [#229a7fccf](https://github.com/fusionps/Arduino_SNMP/commit/229a7fccf9abc7bb67e23f7801329397c20cdc90)

AdamPiorko, 2018-08-01:
It works. Thank you.

closed 2018-09-10 by 0neblock