0neblock/Arduino_SNMP #7: Arduino millis() not able to convert

syntax1269, 2019-05-10:
Trying to add uptime for the arduino. but, i have been unable to get this value to compile.
any help would be appreciated. 
-- code --
snmp.addIntegerHandler(".1.3.6.1.2.1.1.3.0", millis(), false);
-- end code --

give the following
-- error --
/Users/User/Documents/Arduino/test_udp_SNMP/test_udp_SNMP.ino: In function 'void setupSNMP()':
test_udp_SNMP:116: error: invalid conversion from 'long unsigned int' to 'int*' [-fpermissive]
   snmp.addIntegerHandler(".1.3.6.1.2.1.1.3.0", millis(), false);
                                                       ^
In file included from /Users/User/Documents/Arduino/test_udp_SNMP/test_udp_SNMP.ino:6:0:
Arduino_SNMP.h:414: error: initializing argument 2 of 'ValueCallback* SNMPAgent::addIntegerHandler(char*, int*, bool, bool)' [-fpermissive]
 ValueCallback* SNMPAgent::addIntegerHandler(char* oid, int* value, bool isSettable, bool overwritePrefix){
                ^
exit status 1
invalid conversion from 'long unsigned int' to 'int*' [-fpermissive]
-- end error --

0neblock, 2019-05-11:
Hi,

This would be happening because the addIntegerHandler function is expecting a Pointer to an Integer, so it can get the latest value each time it is requested with an external SNMPGET.

The best way to handle an 'uptime' functionality would be to have a global variable that is tracking the uptime, and pass a pointer to that in the call to addIntegerHandler.

The below code snippet should do what you need.

```
... somewhere in global scope:
unsigned long millisecondTracker;

.. in setup():

snmp.addIntegerHandler(".1.3.6.1.2.1.1.3.0", &millisecondTracker, false);

... in loop():

millisecondTracker = millis();
```

syntax1269, 2019-05-13:
Thank you for your suggestion, but getting an error still... tried everything i know to get this to work. Any help would be much appreciated! Other than this little thing everything else seems to work as expected.

--- Start Error ---
Build options changed, rebuilding all
/Users/User/Documents/Arduino/test_udp_SNMP/test_udp_SNMP.ino: In function 'void setupSNMP()':
test_udp_SNMP:102: error: invalid conversion from 'long unsigned int*' to 'int*' [-fpermissive]
 snmp.addIntegerHandler(".1.3.6.1.2.1.1.3.0", &millisecondTracker, false);
 
In file included from /Users/User/Documents/Arduino/test_udp_SNMP/test_udp_SNMP.ino:6:0:
Arduino_SNMP.h:414: error: initializing argument 2 of 'ValueCallback* SNMPAgent::addIntegerHandler(char*, int*, bool, bool)' [-fpermissive]
 ValueCallback* SNMPAgent::addIntegerHandler(char* oid, int* value, bool isSettable, bool overwritePrefix){

exit status 1
invalid conversion from 'long unsigned int*' to 'int*' [-fpermissive]
--- end error ---

syntax1269, 2019-05-13:
Another approach i tried is it create another handler called addLongIntegerHandler, but i just don't know enough of your code and successfully add it in and just got alot more errors.

syntax1269, 2019-05-13:
after alot of errors, i decided to go back to basics.. '
in global i put 
int now = millis();
under void readSensor()
now = millis();
under void setupSNMP()
snmp.addIntegerHandler(".1.3.6.1.2.1.1.3.0", &now, false);

now all works for some reason.. 
getting accurate time ticks.. 
sorry for all the craziness... but, maybe you can improve the existing code to be able to handle Long Int's for someone else that might need to have a use for them in the future.

0neblock, 2019-05-14:
Hi, 

Sorry you are right that it wouldn't natively support a long pointer into that function, My example should have just been an int.

However, depending on which arduino you are using, a long is often the same size as an int.
According to the SNMPv1 spec, all integers are 32 bits, so there would be no difference in how a long and integer are encoded onto the wire.

If I find time i will try to update the library to enable native support of longs.

closed 2019-05-14 by 0neblock