0neblock/Arduino_SNMP #10: SNMPAgent::addFloatHandler, responds with integer in smnpget

wesee-dev, 2019-10-21:
Is it normal for a SNMPAgent::addFloatHandler() send an integer on snmpget?
If so, how can I circumvent this behavior?

0neblock, 2019-10-21:
Hey, unfortunately SNMP, at least versions 1 and 2, doesn’t have a concept of a floating point value. So the addFloatHandler takes the value you provide, and multiplies the value by 10 before sending it out as an integer. 
So the value 19.7F will become 197 integer. 
This is the way most of the devices I saw were handling footing point data. 
There should probably be another variable to that function that allows the multiplier to be changed, to keep more of the data from the original float value, but whatever device is reading this value will have to know how to transform that data back. 

I can quickly add this functionality in if you’d like ?

wesee-dev, 2019-10-22:
I see, one decimal place for my application is good enough but being able to change the multiplier could certainly benefit other users. 
Using string to send the values could be a workaround, could you provide an example of sending string variables if you have the time?
Thanks for the reply

mcpicoli, 2019-10-31:
For sure, using different multipliers are of use in many cases. For example, "power factor" (a dimensionless number that goes from -1 to 1) usually is reported with two decimal places.

closed 2020-11-13 by wesee-dev