0neblock/Arduino_SNMP #12: ESP8266 - wdt reset

Sergio-tix, 2020-02-23:
Hello!
I use ESP-12S [ESP-12S_User_Manual.pdf](https://www.elecrow.com/download/ESP-12S_User_Manual.pdf), Arduino IDE 1.8.11, ESP8266 core for Arduino v.2.3.0, Example - ESP_SNMP.
Co compiles, loads successfully. The console displays:
```
.....
Connected to MyWifi
IP address: 192.168.1.15
```

Next, using iReasoning MIB Browser I am sending a request .1.3.6.1.4.1.5.0 for IP 192.168.1.15 After 5-7 seconds iReasoning gives an error message, 
[image]

And in the console I see this: 

```
......
Connected to GTW
IP address: 192.168.1.15
OID: .1.3.6.1.4.1.5.0

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
```
Settings Arduino IDE

[image]

If you request a nonexistent OID
```
......
Connected to GTW
IP address: 192.168.1.15
OID: .1.3.6.1.4.1.5.5
OID NOT FOUND

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
```
the problem also recurs

P.S. I was looking for information about this error, everywhere they say that the error occurs due to poor nutrition. I power the board from a powerful Power Bank, also has all the necessary capacitors and other elements. But this does not matter, since other projects work fine on this board, the conclusion is that the problem is not in hardware.
Do you see a way to solve the problem?

Sergio-tix, 2020-02-23:
Also, could you give an example of how to work with a data type String? 
I announced
`String sysObjectID = "Sensor10";`
And on the line
`snmp.addStringHandler(".1.3.6.1.2.1.1.2.0", &sysObjectID);`
I see a mistake
`no matching function for call to 'SNMPAgent::addStringHandler(const char [19], char*)'`

0neblock, 2020-02-23:
Hey,

This could be caused by any number of issues, but you’re correct in that it is most likely A firmware problem. 

Could you please post your code, or the relevant parts of it, and I can see if I can identify the issue ?

Thanks.

Sergio-tix, 2020-02-23:
> Привет,
> 
> Это может быть вызвано любым количеством проблем, но вы правы в том, что это, скорее всего, проблема с прошивкой.
> 
> Не могли бы вы опубликовать свой код или соответствующие части, и я смогу определить, смогу ли я определить проблему?
> 
> Благодарю.

I updated ESP8266 core for Arduino v.2.6.3, and now the example works fine. This question can be closed. But still, could you help me with this: [https://github.com/fusionps/Arduino_SNMP/issues/12#issuecomment-590119027](https://github.com/fusionps/Arduino_SNMP/issues/12#issuecomment-590119027)

closed 2020-02-24 by Sergio-tix