0neblock/Arduino_SNMP #14: ESP8266 rebooting (Exception (3)) with long oid request

fcqpl, 2021-02-17:
Hello!
I'm getting ESP8266 resetting while doing snmp get with long oid.
Example OID: `1.3.6.1.2.1.1.5.0.4.1.1.3.4.5.67.2.21.2.23.3.4.5.5.6.2.1222.3`
I'm not using long oids in my application, but I connected LibreNMS in which I wanted to keep charts. During execution of Device Discovery, ESP8266 resets multiple times.
Example OID from Librenms discovery: `1.3.6.1.4.1.8072.1.3.2.3.1.2.6.100.105.115.116.114.111`

Arduino_SNMP should dropping packets with long oid, but i don't know where to implement it.

`Exception (3): epc1=0x40100718 epc2=0x00000000 epc3=0x00000000 excvaddr=0x400082e1 depc=0x00000000`

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x402015b0: ComplexType::addValueToList(BER_CONTAINER*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 394
EXCVADDR: 0x00000004
Memory allocation of 8 bytes failed at 0x402015b0: ComplexType::addValueToList(BER_CONTAINER*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 394

Decoding stack results
0x40201820: ComplexType::fromBuffer(unsigned char*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 516
0x402015a0: ComplexType::addValueToList(BER_CONTAINER*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 593
0x40201810: ComplexType::fromBuffer(unsigned char*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 513
0x402043ec: operator new(unsigned int) at C:\Users\x\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\abi.cpp line 39
0x40201810: ComplexType::fromBuffer(unsigned char*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 513
0x40201810: ComplexType::fromBuffer(unsigned char*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/BER.h line 513
0x40201a2e: SNMPRequest::parseFrom(unsigned char*) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/SNMPRequest.h line 49
0x40202552: SNMPAgent::addIntegerHandler(char*, int*, bool, bool) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/Arduino_SNMP.h line 446
0x40201f75: SNMPAgent::receivePacket(int) at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/Arduino_SNMP.h line 209
0x4020453b: optimistic_yield(uint32_t) at C:\Users\x\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 128
0x40202464: SNMPAgent::loop() at C:\Users\x\Documents\Arduino\libraries\Arduino_SNMP-master/Arduino_SNMP.h line 186
0x40202823: loop() at C:\Users\x\Documents\Arduino\SNMP_agent/SNMP_agent.ino line 66
0x402045fc: loop_wrapper() at C:\Users\x\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 180
```

Best regards! Good library!

fawnlake1, 2021-02-26:
@fcqpl said:

> I'm getting ESP8266 resetting while doing snmp get with long oid.
> Example OID: `1.3.6.1.2.1.1.5.0.4.1.1.3.4.5.67.2.21.2.23.3.4.5.5.6.2.1222.3`
> I'm not using long oids in my application, but I [...]

I am working with the developer 0neblock on this same issue on my ESP32 and wanted to reach out to you with a suggestion that is working at the moment for me. I am using Observium and LibreNMS on testing a solution and I noted that if I disable **UCD-SNMP-MIB** checking within the Observium \ device \ properties \ MIBs section (LibreNMS should have the same section) then the crash stops.

That disables the extended UCD-SNMP-MIB and then I have no issues. **This is not a fix** but it does get me to the next step of reliable operation so I can continue testing and building instead of endless crashing.  Hope it helps in some small way.

[image]

shortbloke, 2021-02-27:
I believe the crashes with long OIDs is caused by a fixed size buffer which only allows max of 50 characters for the OID. I believe an OID can be up to 128. I'll submit a PR for this change shortly.

fawnlake1, 2021-02-27:
@shortbloke I can confirm that making the suggested changes to the BER.h file manually has fixed the issue. Much appreciated!

PR #15 (2021-03-07): Fix for OID longer than 50 characters.
Fixes: #14 

Max OID length was 50 characters, increased to 128 which I believe is the maximum length.
closed 2021-03-07 by 0neblock