0neblock/Arduino_SNMP #21: OIDs greater than 127

sp4c3man, 2021-06-30:
Thanks for your awesome work!
When I'm using OID values <= 127 then FrameFlow can find all OIDs created with snmp.addStringHandler/snmp.addIntegerHandler.
But I have to use greater numbers and when I use an example sketch from another website there is the OID: ".1.3.6.1.4.1.12345.0" used and with this OID my SNMP browser only shows this OID (the first declared) and the others will not be found.

Do you have a solution for this, or have I forgotten something? I am using your actual library, downloaded here.

Thank you for your help!

0neblock, 2021-07-30:
Hi @sp4c3man,

Apologies for the delay. I have seen your issue before, and can confirm it is due to the incorrect way I encode and decode integers.
I saw you investigated a bit on that PR. I'll try to get a working solution up in the next few days.

Thanks, Aidan.

sp4c3man, 2021-08-02:
Hey @0neblock,
thank you! Afterwards my request was a bit shabby, because I did not write something about my hardware etc.
My hardware is an ATMEGA1284p with Bobuino 16 MHz bootloader and my MIB browsers are iReasoning and frameflow, ethernet adapter is a WIZnet 5500. I coded with arduino IDE in C.

In the last days I testet it with an ATMEGA328P (minimum OID declarations, minimum everything) and it did not work at all - maybe for memory reasons, but this microprocessor is not made for this. Then I tried some W5100 and the ENC28J60 with the 1284 but they don't work either.

I wrote with shortbloke in his example thread and he helped me finding some clues why it does/could not work.
If you have not seen: The last error-clue was a request from frameflow and the associated serial output like this:

[image]

But the request from frameflow was ".1.3.6.1.2.1.165.1.1" - after this OID there are some greater ones but he stops after this.

Interesting was this in wireshark:

[image]

.35 is my PC and .50 is the agent - my PC sends a ...199.1.1.1.1.1 request and the agent responses with ...198.1.1.1.1.1 and stops because this OID ist not found.

Maybe you have a solution for this - this would be very cool! But when there is an incompatibility with my hardware, then it is my problem :-)

Thank you very much!

P.S. the delay is no problem at all - we all have jobs an tons of other work to do :-) I'm thankful for every help I can get!

sp4c3man, 2021-08-02:
A little supplement:
Shortblokes example works fine with an ESP8266 (and greater OIDs) - Maybe the compiler automatically corrects "problems" with filetypes etc. and arduino ide does not?

0neblock, 2021-11-10:
Hey, sorry for the delay, this should be fixed by #23

sp4c3man, 2021-11-11:
Hey @0neblock, luckily I had the time to wait, otherwise I would have written to you earlier :-)
Thank you very much for your work! I will test it tomorrow!

closed 2021-11-15 by 0neblock