0neblock/Arduino_SNMP #22: ESP8266 v3.0 breaks library - Use 2.7.4 (or older)

jbuck2005, 2021-08-07:
When using the https://arduino.esp8266.com/stable/package_esp8266com_index.json board file for the esp8266, it is imperative to select version 2.7.4 (or older) for this library to work. Version 3.0 (and newer) causes the following output:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x402023e1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 0190
3fffff70:  4600000a 00ffffff 0100000a 402023e1  
3fffff80:  3ffef568 3ffef588 3ffef648 40202a24  
3fffff90:  40213e58 4600000a 40212dc0 4020158c  
3fffffa0:  3fffdad0 00000000 3ffef784 4020b514  
3fffffb0:  feefeffe feefeffe 3ffe85f0 40100db1  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006df00
~ld
Booting
Ready
```

NourbakhshRad, 2021-09-30:
I have a similar problem,
No solution found?

NourbakhshRad, 2021-11-04:
I discovered that some of boolean functions does not have return
such as:
       SNMPAgent::parsePacket
       SNMPResponse::addResponse
       SNMPResponse::addErrorResponse

I modified these functions (added "return true" at the end of functions) and my problem was solved

0neblock, 2021-11-10:
Hey, sorry for the delay, this should be fixed by #23