0neblock/Arduino_SNMP #28: esp8266 Vscode

Lynxshd, 2021-11-27:
I would like to start by saying thank you for all the work you have done on this library, and for a complete newbie to Arduino and coding micro controllers it was nice to find something that was simple to use,

I am using the snmp_agent with an esp8266 with the arduionIDE but switching to vscode doesn't compile it. I then found this new version of the snmp_agent and was super happy to see that it is in the platformio list, but alas it seems that i can   no longer use this with the esp8266, it seems that there are missing files that stop the file compiling.  is this expected and do you have plans to make it work with the esp8266?

Thanks

Caleb

0neblock, 2021-11-28:
Hi there, This PR: https://github.com/0neblock/Arduino_SNMP/pull/27 has esp8266 compatibility and I've just merged it in. Please try with the latest code.

closed 2021-11-28 by 0neblock

Lynxshd, 2021-11-30:
Thank you so much it worked, I will put it through some testing and let you know if I have any issues. 
Thanks for such a quick response, this is a great lib. :)

Lynxshd, 2021-12-08:
This has been working well and is doing a much better job of passing the info to snmp servers.

Thanks

again