0neblock/Arduino_SNMP #30: No reply to snmpwalk request to .1

filippsudanov, 2021-12-23:
Typically devices respond to 
snmpwalk -v 2c -c public <IP> .1

but the library does not.

0neblock, 2022-01-26:
As the other OID, you should be able to implement this callback to the OID yourself. The library doesn't implement any built-ins.

closed 2022-01-26 by 0neblock

ae-zero, 2022-03-10:
@0neblock 

> As the other OID, you should be able to implement this callback to the OID yourself. The library doesn't implement any built-ins.

Could you give an example?  When you say "implement this callback", do you mean modifying the .cpp/.h files, or do you mean adding a snmp.add<something>Handler to my .ino file?  I've been fighting with trying to get .1 to respond for the better part of a day, and I'm not sure what I'm missing.

@filippsudanov Did you get it working?

Thanks!