0neblock/Arduino_SNMP #33: OID response of .0.0

ae-zero, 2022-03-10:
Hi all,

I'm trying to make my ESP8266 respond with ".0.0" to gets for .1.3.6.1.2.1.2.2.1.22.1 (ifSpecific).  

`snmp.addOIDHandler(".1.3.6.1.2.1.2.2.1.22.1", ".0.0"); // ifSpecific`

I have tried multiple values, such as ".0.0", "0.0", ".0".

When I do an snmpwalk or an snmpget against it, I get a general error:

`[me@myserver ~]$ snmpget -v 2c -c public 192.168.1.7 .1.3.6.1.2.1.2.2.1.22.1
Error in packet
Reason: (genError) A general failure occured
Failed object: IF-MIB::ifSpecific.1`

The corresponding Wireshark output looks like this:

`Simple Network Management Protocol
    version: v2c (1)
    community: public
    data: get-response (2)
        get-response
            request-id: 1573401971
            error-status: genErr (5)
            error-index: 1
            variable-bindings: 1 item
                1.3.6.1.2.1.2.2.1.22.1: Value (Null)
                    Object Name: 1.3.6.1.2.1.2.2.1.22.1 (iso.3.6.1.2.1.2.2.1.22.1)
                    Value (Null)
    [Response To: 1]
    [Time: 0.011983000 seconds]
`

Here is Wireshark output from a box not using this library:

`Simple Network Management Protocol
    version: v2c (1)
    community: public
    data: get-response (2)
        get-response
            request-id: 1702378746
            error-status: noError (0)
            error-index: 0
            variable-bindings: 1 item
                1.3.6.1.2.1.2.2.1.22.1: 0.0 (itu-t.0)
                    Object Name: 1.3.6.1.2.1.2.2.1.22.1 (iso.3.6.1.2.1.2.2.1.22.1)
                    Value (OID): 0.0 (itu-t.0)
    [Response To: 63]
    [Time: 0.035553000 seconds]`

Thanks for your help!

0neblock, 2022-03-16:
Hi @ae-zero, Interesting find with this bug.
As you can see from the OID Encoding method here: https://github.com/0neblock/Arduino_SNMP/blob/8ace14f7c8ac380fe531f3ddfacde8ccb0469cda/src/BEREncode.cpp#L145-L181
I assume that all OID's are prefixed with `.1.3`

I can't remember why I did this originally, but obviously most OID's fit this format.

This can be fixed by changing the encoding to https://docs.microsoft.com/en-us/windows/win32/seccertenroll/about-object-identifier?redirectedfrom=MSDN, which I will add to my list but may not be able to do for a while.

The decoding will also have to be fixed in `BERDecode.cpp`

ae-zero, 2022-03-18:
Thanks for looking into it, @0neblock!  I went in and starting meddling with those two files, but ended up making it much worse :)

Thanks for putting it on your list!

kcscoelho, 2022-06-26:
Hi for all. Sorry for my english.
I have same issue on Mikrotik The Dude app. 
For any reason, First test oid 0.0.
On others systems work Fine, but the Dude is a great app for monitoring.
I thank If verify this on Future.

niltonfreitas0707-max, 2025-09-19:
<details><summary>Details</summary>
<p>

</p>
</details>