0neblock/Arduino_SNMP #34: Lot of libraries missing to compile on a MEGA2560

touffy67, 2022-03-30:
I try to use your code on an Arduino Mega 2560 attached to an ethernet shield.
I cannot compile the project because I have a lot of libraries missing.
In VallueCallbacks.h there is
#include  deque
#include  algorithm
and in ber.h 
#include  utility
#include  vector

where can I find these libraries?

thank you for your help

FairWindCo, 2022-12-02:
Arduino Uno  have same errors. What library need to use this?

ttamiozzo, 2022-12-20:
Same errors here with a mega 2560

0neblock, 2023-01-22:
Hi @touffy67 @FairWindCo @ttamiozzo Sorry for the late reply.
This library should allow it to work on non-ESP versions of Arduino:
https://github.com/mike-matera/ArduinoSTL
I will update the README to indicate this.